PricelessToolkit/CapiBridge #1: Compilation error: 'esp_now_recv_info_t' does not name a type; did you mean 'esp_netif_ip_info_t'?

RonnieLast, 2024-06-17:
Hi, I received the CapiBridge this morning; it looks very good and well designed. 
I flashed ESP1 no problem, ESP2 is encountering an error during compilation. 

```
/home/user_me/Arduino/CapiBridge/ESP1/ESP2.ino:23:23: error: 'esp_now_recv_info_t' does not name a type; did you mean 'esp_netif_ip_info_t'?
 void OnDataRecv(const esp_now_recv_info_t *info, const uint8_t *incomingData, int len) {
                       ^~~~~~~~~~~~~~~~~~~
                       esp_netif_ip_info_t
/home/user_me/Arduino/CapiBridge/ESP1/ESP1.ino: In function 'void parseIncomingPacket(String)':
/home/user_me/Arduino/CapiBridge/ESP1/ESP1.ino:669:3: warning: 'template<unsigned int N> class ArduinoJson::V704PB2::StaticJsonDocument' is deprecated: use JsonDocument instead [-Wdeprecated-declarations]
   StaticJsonDocument<1024> doc; // Adjust size as necessary
   ^~~~~~~~~~~~~~~~~~
In file included from /home/user_me/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:53,
                 from /home/user_me/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /home/user_me/Arduino/CapiBridge/ESP1/ESP1.ino:12:
/home/user_me/Arduino/libraries/ArduinoJson/src/ArduinoJson/compatibility.hpp:63:58: note: declared here
 class ARDUINOJSON_DEPRECATED("use JsonDocument instead") StaticJsonDocument
                                                          ^~~~~~~~~~~~~~~~~~
/home/user_me/Arduino/CapiBridge/ESP1/ESP2.ino: At global scope:
/home/user_me/Arduino/CapiBridge/ESP1/ESP2.ino:11:15: error: redefinition of 'long unsigned int LedStartTime'
 unsigned long LedStartTime = 0;
               ^~~~~~~~~~~~
/home/user_me/Arduino/CapiBridge/ESP1/ESP1.ino:32:15: note: 'long unsigned int LedStartTime' previously defined here
 unsigned long LedStartTime = 0;
               ^~~~~~~~~~~~
/home/user_me/Arduino/CapiBridge/ESP1/ESP2.ino:23:23: error: 'esp_now_recv_info_t' does not name a type; did you mean 'esp_now_peer_info_t'?
 void OnDataRecv(const esp_now_recv_info_t *info, const uint8_t *incomingData, int len) {
                       ^~~~~~~~~~~~~~~~~~~
                       esp_now_peer_info_t
/home/user_me/Arduino/CapiBridge/ESP1/ESP2.ino:35:6: error: redefinition of 'void setup()'
 void setup() {
      ^~~~~
/home/user_me/Arduino/CapiBridge/ESP1/ESP1.ino:117:6: note: 'void setup()' previously defined here
 void setup() {
      ^~~~~
/home/user_me/Arduino/CapiBridge/ESP1/ESP2.ino: In function 'void setup()':
/home/user_me/Arduino/CapiBridge/ESP1/ESP2.ino:51:28: error: invalid conversion from 'void (*)(const int*, const uint8_t*, int)' {aka 'void (*)(const int*, const unsigned char*, int)'} to 'esp_now_recv_cb_t' {aka 'void (*)(const unsigned char*, const unsigned char*, int)'} [-fpermissive]
   esp_now_register_recv_cb(OnDataRecv);
                            ^~~~~~~~~~
In file included from /home/user_me/Arduino/CapiBridge/ESP1/ESP2.ino:2:
/home/user_me/Arduino/hardware/espressif/esp32/tools/sdk/esp32c3/include/esp_wifi/include/esp_now.h:144:54: note:   initializing argument 1 of 'esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t)'
 esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t cb);
                                    ~~~~~~~~~~~~~~~~~~^~
/home/user_me/Arduino/CapiBridge/ESP1/ESP2.ino: At global scope:
/home/user_me/Arduino/CapiBridge/ESP1/ESP2.ino:59:6: error: redefinition of 'void loop()'
 void loop() {
      ^~~~
/home/user_me/Arduino/CapiBridge/ESP1/ESP1.ino:704:6: note: 'void loop()' previously defined here
 void loop() {
      ^~~~

exit status 1

Compilation error: 'esp_now_recv_info_t' does not name a type; did you mean 'esp_netif_ip_info_t'?
```
I don't think it's a library issue, but willing to be wrong on that. 

Arduino IDE is 2.3.2 (flatpak).

PricelessToolkit, 2024-06-17:
Hi, Thank you for supporting my work!. It looks like you've run into a few issues with the compilation. With the latest Arduino IDE on Windows, it compiles without a problem. It's highly likely that the issue you're facing is related to the library on your setup. I recommend checking that all your libraries are up to date and compatible with your current Arduino IDE version.

Hope this helps!

PricelessToolkit, 2024-06-17:
2. The Arduino IDE does not handle multiple firmware files in the same directory well.  /home/user_me/Arduino/CapiBridge/ESP1/ESP2.ino << file placement is not correct.

RonnieLast, 2024-06-17:
Good catch on the directory - I miscreated that alright. 

I have it working now but it was a different problem; I selected "ESP32C3 Dev Module" from "ESP32 Arduino (in Sketchbook)" rather than from "esp32". When I corrected this it compiled and uploaded ok. 

[image]

Another thing to mention is that you might need to tell users to move the UART selector switch back from ESP2 to ESP1 so that they can see the serial output from ESP1 after installing the firmware.

PricelessToolkit, 2024-06-17:
I'm glad it worked, I added point 8 to the instructions "Set the UART switch back to the ESP1 position to see received JSON messages."

closed 2024-06-17 by PricelessToolkit