PricelessToolkit/CapiBridge #3: Capibridge binary_sensors dont auto-detect in MQTT + Homeassistant

genirohtea, 2024-07-27:
I was converting the Mailbox Sensor to use the motion  `m` key and noticed MQTT wasnt correctly publishing to the right topic. The issue comes from: `    String binTopic = String(BINARY_SENSOR_TOPIC) + String(received_json_message.id) + topicSuffix;` not being actually used. Updating the `client.publish(topic.c_str(), value.c_str(), MQTT_RETAIN);` line to select the correct topic based on the type of sensor (regular vs binary), fixes the auto-detection in home-assistant.

PricelessToolkit, 2024-07-28:
Thank you very much! I will check it tomorrow.

PricelessToolkit, 2024-07-28:
You did something like this ??

```c

// Publish the value based on the presence of specific keys
if (doc.containsKey("m") || doc.containsKey("dr") || doc.containsKey("wd") || doc.containsKey("vb")) {
    // For binary sensors like motion, door, window, or vibration, use the binary topic
    client.publish(binTopic.c_str(), value.c_str(), MQTT_RETAIN);
} else {
    // For regular sensors, use the regular topic
    client.publish(topic.c_str(), value.c_str(), MQTT_RETAIN);
}

```

PricelessToolkit, 2024-07-28:
Fixed, there is another bug too, missed

```c
       // auto-discovery for Motion
        "\"payload_on\":\"on\","
        "\"payload_off\":\"off\","

```

closed 2024-07-28 by PricelessToolkit

PricelessToolkit, 2024-07-28:
https://github.com/PricelessToolkit/CapiBridge/commit/6dbeabb0245dc65abf78d720f2d1864c97ce01c0