PricelessToolkit/CapiBridge #6: Issue with MQTT auto discovery with a simple diy pushbutton sender

ppkstat, 2024-11-24:
I have my Capibridge and it works great with the provided examples (such as the ESP32_ESP-NOW_Example.ino). 

Trying to familiarize myself with the protocol, hoping to make some DIY devices for my new gateway I connected a pushbutton to an ESP32 and wrote some code following the guidelines for the receiver side. It works and the Capibridge is getting the following JSON (the leds are also blinking):

`ESP-NOW Message: {"k":"xy","id":"testbutton","bt":"on"}`

Now theoretically this should be enough for the device to be auto discovered in homeassistant but I am getting nothing. Opening the MQTT explorer I am seeing that it publishes on the homeassistant/binary_sensor topic instead the homeassistant/sensor. This is an example:  
`{
  "name": "Button",
  "device_class": "none",
  "icon": "mdi:button",
  "state_topic": "homeassistant/sensor/testbutton/button",
  "unique_id": "testbutton_bt",
  "payload_on": "on",
  "payload_off": "off",
  "device": {
    "identifiers": [
      "testbutton"
    ],
    "name": "testbutton",
    "mdl": "testbutton",
    "mf": "PricelessToolkit"
  }
}
`
Would you be kind enough to tell me if I am doing anything wrong? Perhaps it’s a bug. Is the "device_class": "none" ok for homeassistant?

ppkstat, 2024-11-24:
So happy to be able to help you and contribute! 

I flashed. The mqtt message changed but still nothing on HA: 
`{
  "name": "Button",
  "device_class": "none",
  "icon": "mdi:button",
  "state_topic": "homeassistant/binary_sensor/testbutton/button",
  "unique_id": "testbutton_bt",
  "payload_on": "on",
  "payload_off": "off",
  "device": {
    "identifiers": [
      "testbutton"
    ],
    "name": "testbutton",
    "mdl": "testbutton",
    "mf": "PricelessToolkit"
  }
}`

PricelessToolkit, 2024-11-24:
What you posted is MQTT topic message.

```json

{ "name": "Button", "device_class": "none", "icon": "mdi:button", "state_topic": "homeassistant/binary_sensor/testbutton/button", "unique_id": "testbutton_bt", "payload_on": "on", "payload_off": "off", "device": { "identifiers": [ "testbutton" ], "name": "testbutton", "mdl": "testbutton", "mf": "PricelessToolkit" } }

```
But isn’t the status of that topic either "on" or "off"?

I can’t check it right now since it’s 1 AM. Tomorrow, after my day job, I’ll test it and let you know!

ppkstat, 2024-11-25:
Yes, the status is either "on" or "off". The history graph shows 0 and 1 I don't know if thats normal becauseI am not very familiar with MQTT. Perhaps some pictures might help more:
[image]

[image]

I also flashed another device with your example and it was discovered instantly, just to check if there are any weird issues with my MQTT broker.

PricelessToolkit, 2024-11-25:
Can you add

```c
doc["bt"] = "on";
```

to ESP32_ESP-NOW_Example.ino  "ligne 76"

and test if the button appears or not. 

until tomorrow.
Bay.

ppkstat, 2024-11-25:
I just did. It's not working, so it may be a bug.

PricelessToolkit, 2024-11-25:
Fixed

ppkstat, 2024-11-25:
Yes it works now! So happy to help this great project. I will try to test other data types when I get the chance.