PricelessToolkit/CapiBridge #9: CapiBridge and SX1262 based boards compatibility

Dainius-G, 2025-06-03:
For last few days I have tried to figure out how to add new sensors based Heltec Lora Esp32(V3.1) board.
Tried enabling/disabling crc, SF , Frequency, SyncWord, Preamble combinations etc
With no luck so far 
Any ideas if this at all possible?
thanks.

PricelessToolkit, 2025-06-03:
1. Capibridge has the latest firmware?
2. For SX1262 sync word is 16b, so "0x1424  equivalent of CapiBridge 0x12 "

Connect CapiBridge to a PC via USB and open a serial terminal to see if it receives something.

Dainius-G, 2025-06-04:
I have updated Capibridge today. I can see you added some encryption features, thank you.
back to the issue I have with sx1262 board.
I have ttgo lora 32 v1.3(SX1276 I believe) board and Heltec Lora Esp32(V3.1)() board running identical lora parameters
only difference is I have to use Heltec_unnofficial.h to initialize the sx1262 as I could not get standard Lora.h to initialize sx1262 for some reason. 
TTGO lora 32 v1.3 connects (with syncword 0x12) to CapiBridge (syncWord is 0x12) no problem [https://pastebin.com/yRq6xsJz](url)
Heltec lora 32 v3.1 does not connect (with any syncword 0x12, 0x16b or 0x1424) to  CapiBridge (syncWord is 0x12) [https://pastebin.com/T4EPQW9v](url)

closed 2025-06-10 by PricelessToolkit